DFRobot/DFRobot_DF2301Q #1: Access to raw audio

mikey0000, 2023-04-21:
is there a way to get hold of the audio stream for example to stream to MQTT rasspy if the command isn't recognised?

qsjhyy, 2023-04-21:
I'm sorry, we haven't developed such functionality yet. However, we can enhance the module's capabilities by adding the ability to learn new command words for controlling commands.

closed 2023-04-27 by qsjhyy

JpEncausse, 2024-03-29:
+1 It would be very very valuable if we could have access to the raw audio after a wake word. 
For instance : "Hello bot, (... some audio ...)"

- Because it open the possibility to call LLM like ChatGPT and play back answer
- Because offline trigger to get the audio is very smart compared to always listening that send tons of request.
- Or at least a sample code to record audio just after a wakeword (assuming we do not loss some part) ?

Many thanks for your help.