DFRobot/DFRobot_DF2301Q #5: Library not working with TAMC_GT911 library

GodsonUgo, 2023-12-23:
I have 4.3' TFT TFT touch screen module from Makerfabs https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-4-3-inch.html. I am trying to use the DF2301Q module with the TFT module but it does not work if the TAMC_GT911 touch library (https://github.com/TAMCTec/gt911-arduino) of the TFT module is in the code. The TFT module keeps restarting. If I remove the touch library from the code, the DF2301Q works with the TFT module, and if I remove the DF2301Q library from the code, the touch library works with the TFT module. I am not an expert in Arduino and cannot identify what is causing the problem. Please help me. Thank you in advance.

qsjhyy, 2023-12-26:
At present, I have not obtained more effective information. I can only briefly tell you not to let the communication pins of these two devices conflict. The I2C example of DF2301Q uses the default I2C pins of the master controller, and the UART example uses the pins as explained in the example. These should be modified according to your needs. Please ensure that the relevant pins of the two libraries are configured correctly. The pin configuration for [gt911-arduino](https://github.com/TAMCTec/gt911-arduino) is provided in the example.

qsjhyy, 2024-01-15:
Because there was no follow-up communication for a long time, it was temporarily closed

closed 2024-01-15 by qsjhyy