DFRobot/DFRobot_DF2301Q #6: Additional output for end of wake word listening DF2301Q_UART_MSG_DATA_NOTIFY_WAKEUPEXIT

pinter75, 2024-01-12:
Is it possible to get a CMDID output or similar access when the wake word timeout has occurred?

I see DF2301Q_UART_MSG_DATA_NOTIFY_WAKEUPEXIT in the library, but how can I access it?

It would be great to get a message that the unit has stopped listening. I am using the UART mode.

Thanks!

qsjhyy, 2024-01-15:
Yes, when the module is in UART mode, it does have that functionality. However, it wasn't needed before, so the library doesn't have a dedicated API for it. But the module does automatically prompt that information, and you can actively receive and identify it. Brief information is as follows:
[image]
[image]
[image]

qsjhyy, 2024-04-01:
No follow-up for a long time, so temporarily closed.

closed 2024-04-01 by qsjhyy