DFRobot/DFRobot_DF2301Q #10: Getting I2C error

vonmartins, 2024-05-08:
Greetings, I'm getting this error `requestFrom(): i2cWriteReadNonStop returned Error -1`. When I try the example code for I2C it works but then with my code doesn't. Both codes are more or less the same. I'm using an ESP32-S3 so i use the pins that I want, 5 for clock and 6 for data. I tried using other pins, using pullup resistors, or changing to UART but there's no communication. What it stranges me the most is that sometimes with anotther code, like the example code, it works. But then with another similar code it doesn't. I think it's something about the hardware or even the module itself. Thanks

Here's my code:

```cpp
#include "includes.h"

// ------------------- Voice Module -------------------
DFRobot_DF2301Q_I2C DF2301Q;
// ----------------------------------------------------

void setup()
{

  Serial.begin(115200);
  Wire.begin(I2C_SDA,I2C_SCL);

   while( !( DF2301Q.begin() ) ) {
        Serial.println("Communication with device failed, please check connection");
        delay(3000);
    }
    Serial.println("Voice module enabled!");
    DF2301Q.setVolume(4);
    DF2301Q.setMuteMode(0);
    DF2301Q.setWakeTime(15);
    DF2301Q.playByCMDID(23);
}

void loop() {
    uint8_t CMDID = 0;
    CMDID = DF2301Q.getCMDID();
    if(0 != CMDID) {
        Serial.print("CMDID = ");
        Serial.println(CMDID);
    }
    delay(3000);
}
```

Output:

```
[487111][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[490111][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[493111][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[496111][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
```

qsjhyy, 2024-05-09:
Perhaps you can try using this pin to do an address scan, one by one to troubleshoot the problem.

vance, 2025-07-06:
With a very minimal setup like above, I get

```
E (6089) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (6140) i2c.master: clear bus failed.
E (6140) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (9191) i2c.master: clear bus failed.
E (9191) i2c.master: s_i2c_transaction_start(600): reset hardware failed
E (9192) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (9248) i2c.master: clear bus failed.
E (9248) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (9309) i2c.master: clear bus failed.
E (9309) i2c.master: s_i2c_transaction_start(600): reset hardware failed
```